TheGammaSqueeze/GammaOS #2: RG505

loki666, 2023-05-24:
Are you planning to make this works for the RG505?

rodrif24, 2023-05-24:
I would love it to be ported, if necessary I can serve with testing on my console

vibraniumdroid, 2023-06-02:
I'm also willing to help with any sort of testing

TheGammaSqueeze, 2023-06-03:
Thanks for your interest, I hope to port to the RG505 if someone sends me one for development.

loki666, 2023-06-04:
> Thanks for your interest, I hope to port to the RG505 if someone sends me one for development.

Or you could open source the changes you made to Lineage OS (as you should) so someone could try it on RG505.
Probably only the display parameters needs to be changed for the RG505

TheGammaSqueeze, 2023-06-04:
The issue isn't the open source part, it's how the 505 accepts the image I build from the GSI. The same recompiled GSI image I use for the 405m, causes a boot loop on the 505. The GSi itself is linked in the readme, I'm using this instead of building from source. That link has links to the LineageOS source.

TheGammaSqueeze, 2023-06-04:
In fact, you can install LineageOS easily, without modification by using the GSI.

You will require an unlocked bootloader.
Grab an extracted GSI 19.1x non vndklite.

Extract it so that you have the .img file

Flash using fastboot.

adb reboot fastboot
fastboot flash system systemimagenamehere.img
fastboot format userdata
fastboot reboot

loki666, 2023-06-04:
So andyyan GSIs are working fine on the 405m, but bootloop on the 505 ?

vibraniumdroid, 2023-06-04:
> In fact, you can install LineageOS easily, without modification by using the GSI.
> 
> You will require an unlocked bootloader. Grab an extracted GSI 19.1x non vndklite.
> 
> Extract it so that you have the .img file
> 
> Flash using fastboot.
> 
> adb reboot fastboot fastboot flash system systemimagenamehere.img fastboot format userdata fastboot reboot
> 
> I take no responsibility if something goes wrong, you will need to use the 505 unbricker if you want to go back to stock.
> 
> https://youtube.com/watch?v=A5Ut4W95Z-Q&feature=share8

I've tried this, but regardless of what image I use or whether I try to flash it to system or system_a, I get 

> FAILED (remote: 'Write img fail!')

I guess I'm stuck on rooted stock for the time being

TheGammaSqueeze, 2023-06-11:
I have one on the way now, will hopefully get this ported over soon.

closed 2023-06-11 by TheGammaSqueeze