TheGammaSqueeze/GammaOS #4: RG405m flashing error for vendor_a and system_a

sirkojac, 2023-06-02:
In the log (everything else said OKAY):
Sending sparse 'vendor_a' 1/3 (262140 KB)          OKAY [  9.349s]
Writing 'vendor_a'                                 FAILED (remote: 'No such file or directory')
Sending sparse 'system_a' 1/9 (262140 KB)          OKAY [  7.742s]
Writing 'system_a'                                 FAILED (remote: 'No such file or directory')

Got past the vendor_a 'no such file or directory' error by running 'fastboot create-logical-partition vendor_a 800000'. This was just a guess on size based on the image but it worked on the other hand when I create system_a with the same method (but size set to 3000000) it fails to flash because there isn't enough space to resize. This happens no matter what size I put. I ran 'fastboot getvar' and saw that I was using slot b and "has-slot:vendor:no". Is it possible for the bat file to verify the partitions exist and are the correct size before flashing?

TheGammaSqueeze, 2023-06-03:
Try editing the flashpartitions.bat script.

Change anything that says system_a to system.

Change anything that says vendor_a to vendor.

Save and try running it again.

If that doesn't work then go through the unbricker route.

edwarddoh, 2023-06-07:
Changing system_a to system and vendor_a to vendor actually solved my problem! Thanks @TheGammaSqueeze!!

TheGammaSqueeze, 2023-06-11:
Newest release resolves this going forward: 

Anbernic RG405M - GammaOS / GammaOS Lite Custom Firmware - v1.2.1 - 11th June 2023
----------------------------------------------------------------------------------------------

Download (for both GammaOS and GammaOS Lite):
https://github.com/TheGammaSqueeze/GammaOS/releases/tag/GammaOS_v1.2.1_RG405M

Changelog:
(v1.2.1) Toggle between performance modes by holding down R3+L1+R1. Options are: Max Performance (everything set to max frequency, uses the most battery life), Normal Performance (scales frequency according to load), Power Save (lowers all frequencies to minimum for max battery life, useful for GBA and below). Recommend enabling threaded video in RetroArch for Power Save mode, or change emulators to less demanding ones.
(v1.2.1) Add persistence between reboots for XBOX or NS button layout selection.
(v1.2) Add fix for L2/R2 issue, using my own code https://github.com/TheGammaSqueeze/retrogame_joypad_2_xbox
(v1.2) Presents the physical gamepad as an xbox controller. This may require you to reconfigure your emulators and game controller settings if already on GammaOS.
(v1.2) Back/Home button now functions the same as the stock OS. No more showing recent apps and goes back to home instead on long press.
(v1.2) Toggle xbox button layout by holding L3+L1+R1 for 3 seconds, toggle back by using the same button combo (GammaOS only)
(v1.2) Can use home/back button as a hotkey for RetroArch (only when pressed with another button, otherwise acts as normal home/back button)
(v1.2) Adjust screen brightness by holding home/back button and using the right analog stick UP/DOWN to adjust
(v1.2) Fix flashing issues for those on different partition layouts
(v1.2) Import Anbernic v1.18 firmware brightness scaling fixes
(v1.2) Remove Anbernic stock GPU drivers, Retroid Pocket 3+ perform better and are stable.

Like my work and want to buy me a coffee? Feel free to buy one here: https://ko-fi.com/thegammasqueeze

closed 2023-06-11 by TheGammaSqueeze