TheGammaSqueeze/GammaOS #5: TMNT: Shredders Revenge

jvabbdac, 2023-06-07:
Show netflix screen then close.

TheGammaSqueeze, 2023-06-09:
Is this on GammaOS Lite? 
Does it work on stock firmware?

jvabbdac, 2023-06-10:
Yes, Works fine with stock.

TheGammaSqueeze, 2023-06-11:
I will investigate, thank you.

TheGammaSqueeze, 2023-06-14:
I have found the cause of this. It is related to how Netflix are implementing DRM by using SafetyNet. 
When a custom firmware is flashed after the bootloader on a device is locked, it fails to pass the SafetyNet checks.

However, I was able to get TMNT working but it required a few steps. 

- Use the Magisk app to patch the boot_custom.img that comes with your GammaOS files, then flash this by using fastboot on your PC. Follow this guide on how to do this: https://www.youtube.com/watch?v=0SQWFZQQDwg
- After Magisk is flashed, enable Zygisk in Magisk settings, Enforce DenyList, and Configure Denylist to deny access for Google Play services and TMNT. Reboot.
- Install the following Magisk modules, then reboot
[safetynet-fix-v2.4.0-MOD_1.2.zip](https://github.com/TheGammaSqueeze/GammaOS/files/11744591/safetynet-fix-v2.4.0-MOD_1.2.zip)
.  & 
[Raven.A12L.220705.zip](https://github.com/TheGammaSqueeze/GammaOS/files/11744595/Raven.A12L.220705.zip)

And you should be good to go. 

Video:
https://github.com/TheGammaSqueeze/GammaOS/assets/116582950/1330cd7e-2ac2-49ea-9543-ba3e823b615a

closed 2023-06-14 by TheGammaSqueeze

pmmcrobb, 2023-08-29:
I’ve been having an issue where this game doesn’t save progress whenever I close and reopen it.

CT-Biggles, 2025-01-31:
Would it be possible to look into using a more modern device than the Pixel3A? I'm getting a lot of issues with apps not being compatible with my device in the app store.