TheGammaSqueeze/GammaOS #6: port to retroid pocket 3+

poc-sm, 2023-06-08:
Hey 

Do you think a port can be done for retroid pocket 3+ ?
I think you can flash it with this rom but not sure at all 

the hardware seems the same

Retroid Pocket 3+ Specifications & Features:

    Processor: Unisoc Tiger T618
    CPU: 2*A75@2.0GHz + 6*A55@2.0GHz
    GPU: Mali G52 MC2@850MHz
    DRAM: 4GB LPDDR4x@1866MHz（3733Mbps）
    Storage: 128GB eMMC 5.1
    Touch screen: 4.7 inch, 750x1334@60fps, 450nits 
    OS: Android 11
    Interface: 720p Micro HDMI + 3.5mm Audio Jack + USB TypeC 
    Solid build quality and efficient passive heat dissipation (Metal frame)
    4500mAh battery 

You have a full guide how to root on xda dev https://forum.xda-developers.com/t/rooting-the-retroid-pocket-3.4540153/#post-88395587 

King regards,

TheGammaSqueeze, 2023-06-09:
Without access to a RP3+, I won't know for sure

closed 2023-06-11 by TheGammaSqueeze

Seriousattempts, 2025-05-08:
Did not work for me in my attempt with unlocking the bootloader https://github.com/Seriousattempts/Bootloader_Unlock_Retroid_Pocket_3Plus/blob/main/README.md