TheGammaSqueeze/GammaOS #7: How to build this from sources.

loki666, 2023-06-09:
I'd like to build and play with this for the RG505, but I can't find instructions on how to build this custom ROM from source.

Even if you are starting with prebuilt GSI from someone else, there are obviously some patches you applied to the other images (boot, vbmeta, vendor, userdata ...)

I've found the unified build script and patches repository to build Lineage os GSI from Andy Yan, but your modifications are missing.

Thanks

TheGammaSqueeze, 2023-06-09:
My patches are specific for the 405m, those are in to DTBO partition which defines some hardware parameters, the boot.img for magisk and vendor_boot to include the Black Seraph sd loader mod. The rest are modification via scripts which I include in my modified GSI.

The very first thing you will need to do is at least boot an untouched GSI, you can then work your way from there.

Unlock your 505 bootloader, then get into fastboot.

You will need to use fastboot to flash the Andy Yan GSI build, make sure this is not a VNDKLite build.

command:
fastboot flash system systemimagenamehere.img

You may also need to delete the system_a-cow partition if it exists in your super partition.

Once you get LineageOS booting via a GSI, then you can go ahead with building from source if you desire. Or make modifications as you see fit.

TheGammaSqueeze, 2023-06-09:
As for sources, I literally took Andy Yans build and modified it prebuilt GSI. You can mount my system.img to see what has changed

loki666, 2023-06-09:
Thanks for the details

closed 2023-06-11 by TheGammaSqueeze

sammcj, 2023-12-16:
Any update on being able to build GammaOS for the rg405v device?