TheGammaSqueeze/GammaOS #8: Crashing and rebooting

informadortech, 2023-06-12:
After installing GammaOS 1.1 my RG045M is frequently crashing and rebooting. I flashed again the stock firmware but it continues to crash with it also.

I tried now to install GammaOS 1.2.1 to check if it got fixed but same issue occours. I can't use it more than 15min without it crashed. What can I do? and What info can I provide to help debug?

TheGammaSqueeze, 2023-06-12:
If you have tried flashing the stock firmware and the issue still keeps happening, then this points to being a hardware fault I'm afraid.

I would contact your seller and get a replacement.

informadortech, 2023-06-12:
> If you have tried flashing the stock firmware and the issue still keeps happening, then this points to being a hardware fault I'm afraid.
> 
> I would contact your seller and get a replacement.

It was working fine before unlocking and installing GammaOS the first time.

TheGammaSqueeze, 2023-06-12:
> > If you have tried flashing the stock firmware and the issue still keeps happening, then this points to being a hardware fault I'm afraid.
> > I would contact your seller and get a replacement.
> 
> It was working fine before unlocking and installing GammaOS the first time.

You can try relocking the bootloader after flashing the stock firmware. Do this after a full charge.

Once back in the stock firmware, enable USB debugging. Then issue this command:

adb reboot bootloader

When you're on the Anbernic logo screen, issue the following command:

fastboot flashing lock

Then press the home button on the 405m. It will lock the bootloader.

Then issue the command:
fastboot reboot

See if that helps. I'm afraid I can't help beyond this though.

informadortech, 2023-06-12:
No, I am not using any SD card. Will try to do the process all over again

informadortech, 2023-06-12:
I belive I was on 1.5. I've used the Black-Seraph tool to go back to stock yes.

informadortech, 2023-06-12:
FYI: I booted to fastboot and did a wipe data/factory reset and now thinks look more stable. At least not having reboots. I had to update Daijisho to latest version because the default was crashing after adding paths and press sync button. Let's see how it goes

closed 2023-06-14 by TheGammaSqueeze