TheGammaSqueeze/GammaOS #9: Add flashing script in this repo

poc-sm, 2023-06-12:
Hello,

It could be great if you can add your batch script in this repo 
I'm on GNU/Linux and fork your script to use bash
I think it can be more useful to update if you make change ;-)

I can also make a pr with my script.

TheGammaSqueeze, 2023-06-14:
Hey, it's quite a simple script, it just uses fastboot flash commands, I've just added some extra commands for logging and updating the console window. The main issue is that we need specific drivers for this device on Windows, on Linux I'm not sure if that is required. 

Here are the flashing scripts anyway:
[EraseUserData.bat.txt](https://github.com/TheGammaSqueeze/GammaOS/files/11744687/EraseUserData.bat.txt)
[FlashPartitions.bat.txt](https://github.com/TheGammaSqueeze/GammaOS/files/11744688/FlashPartitions.bat.txt)

poc-sm, 2023-06-14:
just fork your repo and add it
https://github.com/poc-sm/GammaOS/blob/main/FlashPartitions.sh 
I create this issue because if you change something I'll need to report it ;-)

TheGammaSqueeze, 2023-06-14:
Thank you! If you can create a PR and I'll get that merged in, and will include in future releases. LGTM!

TheGammaSqueeze, 2023-06-14:
Actually I have merged it just now, thank you for your contribution.

closed 2023-06-14 by TheGammaSqueeze