TheGammaSqueeze/GammaOS #12: Unable to unlock bootloader

poc-sm, 2023-06-14:
I just receive my rg405m and tryin to install gammaOS
I'm stuck and unable to unlock bootloader 

When I boot into fastboot mode
I go to your website  : https://thegammasqueeze.github.io/subut-rehost/ 
(tryin also https://turtleletortue.github.io/subut-rehost/ )
When I click on connect the pop up open and When I select the device (Fastboot Gadget Associated), it reboot immediately 
[image]

Also tryin to unlock bootloader manually from the device settings, I've got an option "OEM Unlock" but when i tryin to switch it and click on enable it revert back to locked. 

Build Number of the stock ROM : 1.15

Tryin on Arch with chromium and Debian with Chrome 
I think I can do it manually but I don't have the script signidentifier_unlockbootloader.sh to resign the key
https://rootmygalaxy.net/how-to-unlock-bootloader-on-a-unisoc-spreadtrum-device/

poc-sm, 2023-06-14:
Ho gosh
Bootloader is already unlock 
I can boot into fastboot after pluged it
`fastboot reboot fastboot`

poc-sm, 2023-06-14:
Sooooo in fact no ...
I'm able to aces fastbootd but when I try to flashpartitions I've got error 
`command not available on locked devices`

poc-sm, 2023-06-14:
I used my MacOS and chromium and it's worked
Finish on my arch desktop and waiting for the reboot ...

ShalokShalom, 2023-07-16:
Can we open this again? I do run into the same issue, and after the failed attempt to connect to the device, it is impossible to adb to the device:

[image]
[image]

ShalokShalom, 2023-07-20:
Hi. I wonder if the option 'OEM unlocking' does need to be enabled?

poc-sm, 2023-07-20:
Hello
May be you need some udev rules for you user check it out https://wiki.archlinux.org/title/Android_Debug_Bridge 
check 2.2 and 2.3 to add rule

ShalokShalom, 2023-07-20:
We could change the title of this issue, to 'Unable to unlock bootloader on MacOS and Linux'

And then also reflect the state of macOS and Linux support in the README, ideally linking to that issue here. 

Otherwise, people all just run into the same issue, unprepared.

poc-sm, 2023-07-20:
it's indicate in the readme to use google chrome (brave seems based on chrome)
When I unlock bootloader it's worked for me with chromium and MacOS 
So I Think the issue is related to GNU/Linux only

poc-sm, 2023-07-20:
> Can we open this again? I do run into the same issue, and after the failed attempt to connect to the device, it is not possible anymore to boot into fastmode via adb.
> 
> It seems the process has somehow 'revoked the permissions' and I think that's worth looking into it.
> 
> [image] [image] And yes, adb debugging is still enabled on the Ambernic.
> 
> It doesnt help to launch adb as root. I am on Arch Linux. And I used the Brave browser. Thanks!

Can you try to relauch 
`adb kill-server`
`sudo adb start-server`
`sudo adb devices`

I remember something similar when I flashed GammaOS

ShalokShalom, 2023-07-21:
Well, I already unlocked the bootloader in Windows. 

Despite that, I am still interested to get this running in Linux, so if we can work on it at some point, I am definitely trying that. 

For now, I am fine.

inteyes, 2023-07-22:
Just out of curiosity tried to relock/unlock again bootloader on my rg405m with Xubuntu Linux 20.04 + Chromium browser. Have no luck using webpage to do it - ended up with "Unable to claim interface" error, no matter what I tried. 

But I managed to unlock the bootloader using command line by following these instructions - https://www.hovatek.com/forum/thread-32287.html
Only thing that "Home" button should be pressed instead of "Volume down" as in the original GammaOS instructions.

So it's worth a try if there is no access to Mac / Win computers.

ShalokShalom, 2023-07-23:
I am going to try that. 

That's so neat!

fogartys1215, 2023-08-05:
I attempted the install guide with Win10 hosted on Manjaro via VirtualBox.  Unfortunately, once the rg405m rebooted into 'fastboot mode', re-attaching it to the VM would cause the rg405m to reboot itself.

BobbyVans, 2023-08-06:
I have attempted the guide as well. I am on the "Special Website" and when I press connect it doesn't find my device (RG405M). Can anyone help me out here?

ShalokShalom, 2023-08-06:
@BobbyVans 

Did you try this? 
https://github.com/TheGammaSqueeze/GammaOS/issues/12#issuecomment-1646544696

poc-sm, 2023-08-07:
> I have attempted the guide as well. I am on the "Special Website" and when I press connect it doesn't find my device (RG405M). Can anyone help me out here?

Do use use any GNU/Linux distribution ? 
If not please open new issue. 

If yes please provide more explanation 
Your bootloader seems well unlocked so can you retry the flashing procedure

If error occur please provide the output of the log file in the GammaOS folder.

ansemjo, 2023-08-23:
Yup, can confirm that the method from the Hovatek thread linked in https://github.com/TheGammaSqueeze/GammaOS/issues/12#issuecomment-1646544696 works. It is basically the same method that is used in [`main.js`](https://thegammasqueeze.github.io/subut-rehost/js/main.js) on the Chrome "Unlock page" linked above.

The only problem is that a normal `fastboot` binary does not handle the `fastboot flashing unlock_bootloader <signature>` command correctly and I have no idea where you would otherwise get the "modified fastboot", other than in the mega.nz download in that thread.

leiserfg, 2023-09-10:
Could it be that the javascript method does not work in linux because the device is grabbed by `adb` already? 
I'm waiting for my own device, will try stopping the adb service if no one has done it before.   I really don't like that there is no source for the custom `fastboot` of havotek.

ansemjo, 2023-09-10:
When I tried running `adb` commands against the stock ROM manually, I found that `adb devices` did not list the RG405M, while killing the server with `adb kill-server` and restarting it as root with `sudo adb start-server` followed by `adb devices` found the Anbernic.

I'd assume that Chrome's WebUSB API tries to grab the device itself without looking for already running `adb` servers and that this "fails" because the Chrome process doesn't run as root and thus can't find the device.

patrislav1, 2023-10-05:
I have ordered a RG405M (still on the way) and was curious about this issue so I grabbed an old Pixel smartphone and ran the WebUSB fastboot website against it. One thing that immediately popped up is that on Ubuntu 22 LTS, Chromium is run as snap package and seems to have issues with USB device access. So it can enumerate but not open the device ("permission denied"). But it seems to be due to a snap restriction, not to OS level permissions. With default adb & fastboot from Ubuntu package repository, the device file (`/dev/bus/usb/00X/00Y`) is already read/writable by `plugdev` group (via `/lib/udev/rules.d/51-android.rules`). And user 1000 seems to be member of `plugdev` by default. After having installed Chrome via `.deb` [from Google PPA](https://www.ubuntuupdates.org/ppa/google_chrome) it turned out that regular Chrome can indeed open the fastboot USB device. It eventually failed on `oem get_identifier_token` but that is to be expected on my old smartphone I guess? So it looks like there can be multiple levels of permission issues, depending on setup - OS level file permissions (due to missing udev rule and/or `plugdev` membership) and, if Chromium is run as snap/flatpak/AppImage/etc, additional restriction due to sandbox/containerization.

BTW I don't believe that any process or driver from the adb suite is blocking the device file, since `lsusb -t` doesn't show any driver using the device and neither does `lsof /dev/bus/usb/00X/00Y`. Also when running `strace` on `fastboot` it obviously just opens one device file after another from userspace, not different from what a browser running WebUSB should be doing.

Still waiting patiently for my RG405M to arrive and curious how it will turn out, but maybe this little infodump can help other people in the meantime.

patrislav1, 2023-10-14:
Today I received my RG405M. Bummer, I get the same error ("Unable to claim interface") even when running the site on original Chrome from Google PPA.

patrislav1, 2023-10-15:
I have ported the WebUSB code to Python and successfully used it to unlock my RG405M on Linux!
So neither Chrome/Chromium nor the patched fastboot binary is necessary any more.

https://github.com/patrislav1/unisoc-unlock

ShalokShalom, 2023-10-15:
Maybe add that to the README?

closed 2024-01-16 by poc-sm

Shaneadam88, 2024-11-24:
> I just receive my rg405m and tryin to install gammaOS
>
> I'm stuck and unable to unlock bootloader
>
>
>
> When I boot into fastboot mode
>
> I go to your website  : [...]

I'm stuck on this screen can anyone please help me I try to boot into fastboot but keeps coming back to this screen. It's the rg406v I got it from a friend with no sd card just internal 128gb storage. It was working fine and than all of a sudden I got this message and it won't go past this screen! Can anyone please help me? [image][image]

ShalokShalom, 2024-11-25:
You are in the wrong repo (and thread)

https://github.com/TheGammaSqueeze/Bootloader_Unlock_Anbernic_T820