TheGammaSqueeze/GammaOS #13: Update to 1.2.1

jvabbdac, 2023-06-15:
I Try update from 1.1. to 1.2.1. The processes took like 10 sec. Don't show number "0", show "9009". Log : 'fastboot' is not recognized as an internal command'

poc-sm, 2023-06-16:
Seems fastboot command is not installed/recognized 
Can you install / reinstall fastboot
https://github.com/K3V1991/ADB-and-FastbootPlusPlus
have a look at 2m35 
https://www.youtube.com/watch?v=PdE9vNRRjUI

jvabbdac, 2023-06-16:
> Seems fastboot command is not installed/recognized Can you install / reinstall fastboot https://github.com/K3V1991/ADB-and-FastbootPlusPlus have a look at 2m35 https://www.youtube.com/watch?v=PdE9vNRRjUI

its done! I uninstall and reinstall and worked. Thx!

closed 2023-06-19 by TheGammaSqueeze