TheGammaSqueeze/GammaOS #14: Bootloop after reboot to recovery

gibaBR, 2023-06-20:
Hi, first, thank you for your great work on GammaOS.

Today, I was preparing to upgrade my RG405 GammaOS to version 1.3 (was on 1.2). Then, follwing the instructions, I rebooted it to recovery ( hold down the power button and choose the Restart option, then choose the Recovery option.). Then it started bootlooping on Ambernic Logo and the usual messages are shown: 
INFO: LOCK FLAG IS: UNLOCK!!
Warning: Lock Flag  is: Unlock, Skip Verify!!!

I recorded a video: https://www.youtube.com/watch?v=FJmCkqdZDNo

gibaBR, 2023-06-20:
I had to flash back the original firmware using Black-Seraphs. 
Now its working again but I'll have to configure everything again.

TheGammaSqueeze, 2023-06-20:
Hi, sorry about this. This may be related to your SD card, if you see this issue again, try ejecting your SD card.

gibaBR, 2023-06-20:
Nice! Never thought it could be the SD card. Next time, should I remove the SD card before try to update?

TheGammaSqueeze, 2023-06-20:
Yes that would be the workaround

closed 2023-06-22 by gibaBR

jdtgr, 2023-09-21:
Hi, I have an rg505 same boot loop owing stop vibrating and screen flashing. 
There was no sad card inside mine when it started. 
Any help with re install firmware using a Mac? Unfortunately I have available