TheGammaSqueeze/GammaOS #17: Daijisho and Retroarch settings completely erased after an OS reboot

eduardopatricio, 2023-06-23:
After a fresh install of GammaOS v1.3.1 (no issues during installation), I added some paths to Daijisho and synced them. I also made my usual Retroarch adjustments (UI, hotkeys, video etc.). 
Upon rebooting, Daijisho showed no paths configured, no extra platforms added, nothing, just like it was during the fresh install. Same thing with Retroarch. 
I tried to make modifications and reboot to confirm the issue 3 more times. Everything was reset every time. 

(FYI OS settings, such as Google account, security settings etc. persist. So do additional installed apps. I tried configuring Duckstation and rebooting the device, the Duckstation settings persisted. )

eduardopatricio, 2023-06-23:
I figured this had to be an isolated bug, a read and write issue unique to my device.
I did a factory reset and the issue seems to be gone.

poc-sm, 2023-06-23:
I think is isolate 
I did an upgrade and fresh reinstall in v1.3.1. I don't have any issue like you describe 
glad to ear a factory reset resolve your issue

ale296, 2023-06-23:
Does "fresh reinstall" mean you have to reflash to stock and then install Gammaos?

loki666, 2023-06-23:
> I figured this had to be an isolated bug, a read and write issue unique to my device.
> I did a factory reset and the issue seems to be gone. 

Having the same issue.
Did you reflash the partitions with the script?

loki666, 2023-06-24:
I had to uninstall and reinstall RetroArch, and clear the data of Daijisho
Now it's working properly, I guess some filesystem persmission issue

eduardopatricio, 2023-06-24:
> > I figured this had to be an isolated bug, a read and write issue unique to my device.
> > I did a factory reset and the issue seems to be gone.
> 
> Having the same issue. Did you reflash the partitions with the script?

I didn't go back to the scripts, no. I just chose "Wipe data/factory reset" from recovery. 

Now I remember that when I first installed the OS, I installed the latest version of Retroarch (64) from the website before configuring anything. Maybe that messed things up (writing permissions or something)?

loki666, 2023-06-24:
> Now I remember that when I first installed the OS, I installed the latest version of Retroarch (64) from the website before configuring anything. Maybe that messed things up (writing permissions or something)?

no, I didn't update RetroArch the first time, and I had the issue

TheGammaSqueeze, 2023-06-24:
OK so based on your description of the issue, this might be provisioning flag that should be set not actually being set. 

This is just a check for an empty file in /data/data/setupcomplete, which is created after first boot, but only once all customization has been completed, including setting RetroArch config and cores, along with Daijisho. 

If this isn't set for some reason, then on every reboot, the provisioning script I set up will continue to try and reprovision the two apps. 

I think I know what is causing this, and I believe this will only happen on GammaOS and not Lite. I'll get this fixed, but will mean a longer first boot to ensure that all flags are correctly set and all packages have been extracted first before allowing the user to proceed with device setup.

TheGammaSqueeze, 2023-06-24:
If you want to create this flag yourself, just use a shell as root to run the following command:
touch /data/data/setupcomplete

loki666, 2023-06-24:
> If you want to create this flag yourself, just use a shell as root to run the following command:
> touch /data/data/setupcomplete

setupcomplete file was already there.
Tried touch again but it still restore ROMs and config on boot

TheGammaSqueeze, 2023-06-24:
Hmm that's odd, it should not be provisioning if the file is there. I wonder if your /data is being mounted after the script starts hence why this condition might fail for you. That doesn't seem right though. 

Can you try deleting that file, reboot your device, let it sit for 5 minutes then see if the file shows up?

loki666, 2023-06-24:
yup file is comming back

TheGammaSqueeze, 2023-06-24:
Now can you delete the retroarch app and see if it reappears after reboot?

loki666, 2023-06-24:
1.3.2 seems to have fixed it
but now I looks like I have touch inputs issues...
Will open a new issue if it keeps happening

closed 2023-06-24 by TheGammaSqueeze

TheGammaSqueeze, 2023-06-25:
This is now fixed permanently in the 1.3.3 release.
The root cause was that when screen lock with password or pin etc is used, the /data partition is encrypted. 

The provisioning script doesn't see the setupcomplete file as a result hence the reprovisioning.

I have now moved this to use the built in Android provisioning flag, which works outside of any file system and is stored within the settings database.

loki666, 2023-06-25:
I was about to complain about the boot time, but 1.3.3 fixed it
thanks for your work