TheGammaSqueeze/GammaOS #19: touchscreen input issues

loki666, 2023-06-24:
Since v1.3.2 I'm having some "ghost" touchscreen inputs (rendering the device RG505 unusable)
When I touch the screen, it seems to register random touch in other parts of the screen

TheGammaSqueeze, 2023-06-24:
I have no idea what is causing this.

Does downgrading to an older GammaOS build fix it?

TheGammaSqueeze, 2023-06-24:
Oh hang on, are you using GammaOS Lite? Part of of the provisioning script is to emulate a tab keypress and an enter keypress.

If your device isn't being provisioned properly then it will do this step while booted up.

If you are on GammaOS Lite with this issue, I suggest a factory reset or run the EraseUserData script which in fastbootd mode.

loki666, 2023-06-24:
I'll try a factory reset and downgrade to see if it fixes the issue but I'm not on Lite version. (But I did try Lite 1.3.1 before flashing 1.3.2 non Lite, and I did run the Erase script after flashing 1.3.2)

loki666, 2023-06-24:
Ok so
Factory reset via recovery didn't help.
Erase script did the trick until I set the System navigation to gesture mode (no system button for back/home etc)

As soon as I set this device UI is almost frozen. I manage to set it back to 2 or 3 buttons.

Didn't experienced this with v1.3.1

TheGammaSqueeze, 2023-06-24:
Sorry to hear you're still having issues...

I did try it for myself on v1.3.2 with no issues
https://youtube.com/watch?v=CRixaCDzVhk&feature=share7

All I can suggest is that maybe reinstall the stock firmware via the unbricker method and see if you still get issues after this, either on stock os or GammaOS 

https://github.com/TheGammaSqueeze/GammaOS#help-my-device-is-no-longer-booting-or-i-want-to-go-back-to-stock-os

loki666, 2023-06-24:
Weird...

I was enabling gesture navigation via Settings -> Accessibility -> System controls

I just enabled it from System -> Gestures and it seems to works so far...

But via Accessibility was working on v1.3.1...

And in my About device menu, for the build number in don't have 'GammaOS 1.3.2'... I have 'SP2A.220505.008' ???

TheGammaSqueeze, 2023-06-24:
Ah yeah thats the magisk module enabled which spoofs the pixel 3a. You can disable this and reboot. You'll see the true version number there.

Otherwise, if its working now then I'll close this specific issue and review the accessibility option. However I never made any changes in this area, so this is odd.

closed 2023-06-24 by TheGammaSqueeze