TheGammaSqueeze/GammaOS #21: Audio latency/lag (audio buffer size)

MetroidFusion, 2023-06-26:
Hi

On stock firmware, both the RG405M and RG505 suffers from audio latency/lag. This issue is caused by the audio buffer size in the Android firmware which is too large. It's set to 960, when most modern devices and phones have it set to around 128-256. The fix for the audio latency lag issue should be quite simple:

You will have to change a few parameters in the following file:

/vendor/etc/audio_pcm.xml

Open audio_pcm.xml in a text editor, and replace all instances of the number '960' with '256' (or even lower). This will fix the issue.

The above method has been tested and is working on the Retroid Pocket 3+ (which also has the T618 chip). The fix has subsequently been integrated in the stock firmware for this device via OTA.

Would you consider implementing this fix to GammaOS?

/M

TheGammaSqueeze, 2023-06-26:
Hi, 

I will take a look at this and if it all works well, will incorporate into the next release.

Thank you.

MetroidFusion, 2023-06-26:
> Hi,
> 
> I will take a look at this and if it all works well, will incorporate into the next release.
> 
> Thank you.

That sounds awesome. If you want to see how much lowering the buffer size improves overall audio latency, you can use this app:

https://superpowered.com/latency

On the Retroid Pocket 3+, lowering the buffer size reduces audio latency significantly. It makes rhythm games like Patapon actually playable :)

/M

TheGammaSqueeze, 2023-07-04:
In the interim, I have created a Magisk module which will set this value to 160 blocks. 
[SPRDAudioBufferFix.zip](https://github.com/TheGammaSqueeze/GammaOS/files/11952300/SPRDAudioBufferFix.zip)

In my testing, it works very well. 

Unfortunately the latency app you mentioned doesn't work here as there doesn't appear to be a microphone on this device, however I have tested with this app and can see that audio begins at the 0ms mark: https://play.google.com/store/apps/details?id=barum.life.wirelessearbudslatencytester

I am testing with the following RetroArch settings
[image]

Seems to be working great from my testing.

MetroidFusion, 2023-07-05:
Great work! As for the latency app I mentioned, I do know that the RG505 has a microphone working on stock firmware. I have used both the aforementioned latency app and a 'sound recording app' on the RG505 successfully :)

Will you implement the Magisk module in future releases of GammaOS?

/M

TheGammaSqueeze, 2023-07-05:
Ah yes just tested on the 505, it does indeed have a microphone!

We get a reduction from 400+ ms to around 130ms, this is setting the buffer size to 64 (from 960).

I have updated the Magisk module with the new values, and have also added Bluetooth apt-x codec support to further reduce latency via those devices too.

[SPRDAudioBufferFix2.zip](https://github.com/TheGammaSqueeze/GammaOS/files/11962585/SPRDAudioBufferFix2.zip)

This is being tested at the moment, and it seems to show promising results for other users too.

Once enough testing is done, I'll incorporate the changes into the firmware permanently, without the need of the Magisk module.

MetroidFusion, 2023-07-06:
> Ah yes just tested on the 505, it does indeed have a microphone!
>
> We get a reduction from 400+ ms to around 130ms, this is setting the buffer size to 64 (from 960).
>
> I have updated the Magisk module [...]

That sounds amazing - the reduced latency will make a big difference in regards to the overall experience of the RG505 and RG405M as emulation devices.

Thank you very much, I'm looking forward to the next release of GammaOS :)

/M

TheGammaSqueeze, 2023-07-07:
Have now managed to get the latency down to 88ms.

MetroidFusion, 2023-07-07:
> [image]
> 
> Have now managed to get the latency down to 88ms. [SPRDAudioBufferFix3.zip](https://github.com/TheGammaSqueeze/GammaOS/files/11984177/SPRDAudioBufferFix3.zip)

Wow, that's impressive. Probably the lowest audio latency I have seen on any Android device.

TheGammaSqueeze, 2023-07-13:
[SPRDAudioBufferFix4.zip](https://github.com/TheGammaSqueeze/GammaOS/files/12044936/SPRDAudioBufferFix4.zip)

Adding a newer version of the fix as a magisk module, so it can be used with other T618 devices

closed 2023-07-13 by TheGammaSqueeze

wansti, 2023-08-02:
Just tested [SPRDAudioBufferFix4_anbernic.zip](https://github.com/TheGammaSqueeze/GammaOS/files/12045133/SPRDAudioBufferFix4_anbernic.zip) on my RG405M and it doesn't seem to make a difference. Is there anything else I need to do besides installing the Magisk module and rebooting?

therealgorgan, 2024-02-07:
> Just tested [SPRDAudioBufferFix4_anbernic.zip](https://github.com/TheGammaSqueeze/GammaOS/files/12045133/SPRDAudioBufferFix4_anbernic.zip) on my RG405M and it doesn't seem to make a difference. Is there anything else I need to do besides installing the Magisk module and rebooting?

Same issue here. I just got my RG405M last week from Anbernic Direct on TikTok shop and flashed GammaOS today just to fix the bluetooth audio issue, but the Magisk package doesn't seem to resolve the issue. 

Is there a way I can try some of the other settings mentioned earlier in the discussion? I guess we'd have to make a new Magisk module package, but I don't know the first thing.

virtualcomet, 2024-02-12:
Hi! I am so confused, I have the latest Gamma OS Lite installed on my RG405M. In RetroArch audio settings, it says my "block frames" is 960. I cannot change it. The post here says to change the parents in the folder "/vendor/etc/audio_pcm.xml". Where do I find this folder? I've searched everywhere in the internal storage of my RG405M and I don't see any folders named "vendor" at all. Please help me if you can, thank you so much!

therealgorgan, 2024-02-13:
I couldn't figure out a way to change the file directly but I tried using the Magisk module that the developer posted above, you can simply extract the contents and then change whatever you need in there and then create a new zip of it to make a new Magisk module. 

Like you, the Bluetooth issue affects me and I still haven't figured out how to resolve it on my new RG405M, even trying different variables as mentioned by other comments

virtualcomet, 2024-02-13:
> I couldn't figure out a way to change the file directly but I tried using the Magisk module that the developer posted above, you can simply extract the contents and then change whatever you need in [...]

I’m very curious because you said you did the Magisk method. Can you scroll up and look at Gamma’s screenshot of his audio output settings? You see where his says “160” at that bottom for “blocked frames”? What does yours say? Mine says 960.

PacoRm10, 2024-04-14:
Can you show me how to install this? I just updated to GammaOs on my RG405V and it didn't fix the latency issue. I downloaded the magisk module but I can't install it (or any other) from the magisk app. I tap on install from storage but I can't select any file.

virtualcomet, 2024-04-17:
> Can you show me how to install this? I just updated to GammaOs on my RG405V and it didn't fix the latency issue. I downloaded the magisk module but I can't install it (or any other) from the magisk app. I tap on install from storage but I can't select any file.

Are you talking about REGULAR audio latency or WIRELESS BLUETOOTH audio latency? One is able to fix the BLUETOOTH latency because it must be a hardware thing. I spent weeks trying it make it work because my wife just bought me $350 Sony earbuds and I wanted to use it for the device. I even bought adapters for it too but ultimately gave up and bought a Tin HiFi T4 Plus in-ear monitor earbuds for $120 and I'm so happy with it.