TheGammaSqueeze/GammaOS #22: Quick Settings tiles for Performance and Xbox buttons

loki666, 2023-06-26:
Like the Stock OS.
I can help developping this, but I need to know how to interface with both services.

(If only your changes were open source...)

TheGammaSqueeze, 2023-06-26:
Appreciate any help with this

Performance and xbox toggle code is in here, it references some scripts that I've bundled into the os image:
https://github.com/TheGammaSqueeze/retrogame_joypad_2_xbox/blob/main/retrogamepad2xbox.c

Be warned, it's messy!

TheGammaSqueeze, 2023-06-26:
Feel free to optimize. It works fine for our purposes, and needs to be responsive enough to register inputs. Util is actually pretty low, check using top command.

You are just building an apk correct? Sign with any key. This image was not built from source, but rather modified an existing GSI. Nothing stopping me from building from source, but it takes forever and my changes are minimal anyway.

This is an erofs image, read only. Mount:
mount -t auto -o ro .img mountfolder

Now if you want to modify, you will need to copy the mountfolder while preserving all selinux and other attributes.

cp -pdavr mountfolder yourworkingfolder

To build from your working folder, cd into the mountfolder inside yourworkingfolder.

Use 
mkfs.erofs -z LZ4HC 

to do this, ensuring you are inside the actual yourworkingfolder/mountfolder/ when building the image, ensure the image is created outside of this folder.

Good luck.

loki666, 2023-06-26:
Mmmh ok...
I think I was missing the erofs driver.

TheGammaSqueeze, 2023-07-02:
Closing this for now, happy to reopen if there are aby updates.

closed 2023-07-02 by TheGammaSqueeze