TheGammaSqueeze/GammaOS #23: How to recalibrate joysticks? Drift in 3 directions..

kartsnkittiesGF, 2023-06-27:
Experiencing stick drift left, right, and down. Is there a way to recalibrate the joysticks? Your stock calibrations might be a little too "tight". Pretty cool otherwise though

TheGammaSqueeze, 2023-06-27:
That doesn't sound like a GammaOS problem, but more like when yhe device is booting, the kernel has calibrated your sticks wrong, maybe they were being pushed  during boot.

Try turning off your device, then move the sticks in a circular motion a few times. Then turn back on, don't touch the sticks until you're in GammaOS.

kartsnkittiesGF, 2023-06-28:
I've already tried this, the issue persists. And it's not in one direction, it's in all directions. It will drift left, then right, then back, then forward, all in one session. I hadn't had this issue before I switched TO GammaOS. Anyhow, is there a way I can increase the dead zone on my own? This is happening in android apps, so it's not like I can rely on an emulator setting, I need to increase the overall dead zone on the device.

TheGammaSqueeze, 2023-06-28:
There isn't a way to increase the dead zone I'm afraid.
I would suggest switching back to the stock os to confirm if the issue persists. Can you try this?

closed 2023-07-01 by TheGammaSqueeze