TheGammaSqueeze/GammaOS #24: Cannot open any ps2 games, retroarch freezes on start

neamtua, 2023-06-29:
I've just installed gammaos last night on my RG405M and I've been trying to run the games that came with the sdcard in the package. For instance god of war 2. It starts and there's just a black screen and then after a while i might get a message like retroarch isn't responding.

I know for sure I could play these on the stock OS.

Any tips?

gibaBR, 2023-06-29:
Verify of the core for ps2 in retroarch is installed.

neamtua, 2023-06-29:
Sorry, I'm kind of new at this. Which one is the PS2 core? There are a few Sony Playstation ones there but nothing Playstation 2.

gibaBR, 2023-06-29:
It would be PCSX2, BUT you might want to use a standalone app
called  AetherSX2 (download it from google store).
---
Gilberto Júnior
http://raspberrybrasil.blogspot.com.br/
http://www.flickr.com/photos/gibajr/

[image: Mailtrack]
<https://mailtrack.io?utm_source=gmail&utm_medium=signature&utm_campaign=signaturevirality11&>
Sender
notified by
Mailtrack
<https://mailtrack.io?utm_source=gmail&utm_medium=signature&utm_campaign=signaturevirality11&>
29/06/23,
19:09:07

TheGammaSqueeze, 2023-07-02:
Not a GammaOS issue, please use AetherSX2 for your emulator, not RetroArch.

closed 2023-07-02 by TheGammaSqueeze