TheGammaSqueeze/GammaOS #25: God of war doesn't work for me like it did for you

rodrif24, 2023-06-30:
I'm using an rg505 and in your video the game works very well, while in mine with gammaos 1.3.3 it doesn't work very well, I tried various settings.

Actualmente : 
version 13930 v.1.5-3668 
Bios : europe v2.00 2004
Gow of war Pal
Default settings with vulkan , EE cycle rate -1
cycle skip 2
What version of the emulator do you have, and what options do you recommend?

I made a video to show you how it works 

thank you so much for the system

gibaBR, 2023-06-30:
I always use this guide:
https://www.youtube.com/watch?v=PMw0A-iFU0Q&list=PLN5ZVMX1XJzzAazELXt-CPkQuVjRI4oF1
to configure my device.
If you don`t want to follow everything, there is a chapter dedicated to PS2
emulation.

---
Gilberto Júnior
http://raspberrybrasil.blogspot.com.br/
http://www.flickr.com/photos/gibajr/

[image: Mailtrack]
<https://mailtrack.io?utm_source=gmail&utm_medium=signature&utm_campaign=signaturevirality11&>
Sender
notified by
Mailtrack
<https://mailtrack.io?utm_source=gmail&utm_medium=signature&utm_campaign=signaturevirality11&>
30/06/23,
07:15:46

TheGammaSqueeze, 2023-07-02:
Closing this issue.

My settings are at the beginning of this video: https://youtube.com/watch?v=cItjMg22S74&feature=share8

Make sure you are in max performance mode, I'm using the latest Aethersx2 build (the one with ads, but we block those on GammaOS anyway).

closed 2023-07-02 by TheGammaSqueeze