TheGammaSqueeze/GammaOS #26: "Select" and "Start" cannot be used to quit RetroArch.

DieHardMan5x, 2023-07-01:
"Select" and "Start" cannot be used to quit RetroArch. Even with Xbox Mode disabled it does not work. Other hotkeys together with "Select" work strangely enough. 
I have tried all the settings but nothing has helped. "Confirm quit" is enabled.

It works on the stock firmware however.

[image]
[image]
[image]
[image]

Sometimes shortly after starting a rom I get the correct confirm message to appear.

[image]

eduardopatricio, 2023-07-01:
I believe your problem might be your current Retroarch settings. I see you
have "Hotkey enable" set to the 'select' key (109). That means all your
hotkeys will be combos by default ("select" + "some-other-key").
On top of that, you have the "quit controller combo". So, you're mapping
"select" to 2 different things.

If you use select as a hotkey enabler for other things, I'd keep "select"
as "hotkey enabler", remove the "quit (controller combo)" option and set
the "start" key as "Quit".
In any case, even with your current problematic double mapping, pressing
select and start precisely together simultaneously should work to quit the
game. (I've double checked your configuration on my device here).

I hope this helps.
Cheers,
E.

*Eduardo Patricio, PhD*

Sound Designer / Lecturer at Bydgoszcz Music Academy
https://linktr.ee/eduardopatricio

DieHardMan5x, 2023-07-01:
Thanks for the help, if I use "Start" as a hotkey for "Quit" then everything works fine. There my memory played a trick on me, the issue can be closed.

closed 2023-07-01 by DieHardMan5x