TheGammaSqueeze/GammaOS #27: Slow boot when Gesture Navigation enabled

loki666, 2023-07-07:
When I enable Gesture Navigation (instead of 3 buttons) boot is very slow (System UI not responding and I need to close it)

As soon as I revert to the button navigation issue disappears and boot is fast again.

TheGammaSqueeze, 2023-07-07:
I'm unable to replicate this issue, let me know if this is how you set yours up

https://github.com/TheGammaSqueeze/GammaOS/assets/116582950/a16f161b-e1b4-4337-84ec-6f03ab6b34a9

loki666, 2023-07-07:
It was on RG505, but yes.
Already reflash + factory reset... As soon as I enable Gesture Navigation, boot is acting weird... slow / stuck... sometime with an old styled reboot dialog prompt on the screen. No Gamma logo

TheGammaSqueeze, 2023-07-07:
Did you set the screen lock by any chance? I will try on the 505 shortly

loki666, 2023-07-08:
Setting a screen lock is part of my typical device setup, but I thought that could be the problem, so after my last reflash I didn't setup the screen lock.

Only the gesture navigation...

loki666, 2023-07-08:
here is the boot with button navigation
https://github.com/TheGammaSqueeze/GammaOS/assets/400014/387a1cb3-9afd-4f1f-9037-9b1ea264dba9

here is the boot with gesture navigation
https://github.com/TheGammaSqueeze/GammaOS/assets/400014/7831804d-c399-4f5e-ba97-0900135dba0f

TheGammaSqueeze, 2023-07-08:
Ah I see. I'm still not able to replicate on the 505.

I think I will require the adb logcat output when you are experiencing issues.

Connect your 505 via usb and run this command:

adb logcat –d > filename.txt

This command will extract the logcat information from the connected device and redirects the output to a file on the PC.

You can send this log to me to review.

loki666, 2023-07-09:
https://github.com/TheGammaSqueeze/GammaOS/assets/400014/d6eeca9a-752f-4c51-8cfa-42e4f3caa9ea

[boot_gesture.txt](https://github.com/TheGammaSqueeze/GammaOS/files/11994817/boot_gesture.txt)
[boot_gesture2.txt](https://github.com/TheGammaSqueeze/GammaOS/files/11994818/boot_gesture2.txt)
[boot_gesture3.txt](https://github.com/TheGammaSqueeze/GammaOS/files/11994819/boot_gesture3.txt)

boot_gesture.txt was captured when the device booted, but then logcat stopped, but device was still starting up, so I created 2 others

TheGammaSqueeze, 2023-07-19:
Sorry I've not been able to follow up on this, could you by any chance disable all Magisk modules and disable Zygisk in Magisk settings? Let me know if that helps.

closed 2023-07-24 by TheGammaSqueeze