TheGammaSqueeze/GammaOS #28: Permission is denied when using frontend to launch game with standalone emulator

inteyes, 2023-07-08:
Hello!

There seems to be something that is preventing roms from running through standalone emulators with daijisho:
```
07-08 17:38:15.040  3413  3543 I redream : opening /storage/74E9-F089/Games/dreamcast/Power Stone 2.chd
07-08 17:38:15.049  2072  2572 E MediaProvider: Permission to access file: /storage/74E9-F089/Games/dreamcast/Power Stone 2.chd is denied
07-08 17:38:15.049  3413  3543 W redream : failed to open /storage/74E9-F089/Games/dreamcast/Power Stone 2.chd
```
There are no issues with running the games from the emulators itself though.
I'm not quiet sure if this is directly related to GammaOS, but there is no such problem on stock firmware.

Thanks!

[logcat.zip](https://github.com/TheGammaSqueeze/GammaOS/files/11990778/logcat.zip)

TheGammaSqueeze, 2023-07-08:
Hi there, I think this isn't a GammaOS issue specifically, any newer version of Android will exhibit this same behavior due to how access to files are now define as of Android 12+ with scoped storage. 

Notice how Daijisho tries to send the following file path but it fails to open in Redream:
07-08 21:45:22.795 13861 10753 I redream : opening **/storage/6AEB-F59A/dreamcast/Sonic Adventure (U).cdi**
07-08 21:45:22.799  2086  7870 E MediaProvider: Permission to access file: **/storage/6AEB-F59A/dreamcast/Sonic Adventure (U).cdi is denied**
07-08 21:45:22.799 13861 10753 W redream : **failed to open /storage/6AEB-F59A/dreamcast/Sonic Adventure (U).cdi**

Now this is what the file path looks like when redream opens the file directly:
07-08 21:46:06.777 13861 10973 I redream : opening **content://com.android.externalstorage.documents/tree/6AEB-F59A%3Adreamcast/document/6AEB-F59A%3Adreamcast%2FSonic%20Adventure%20(U).cdi** 

If we can do some sort of transform on the string in the Daijisho settings when it passes it through to redream, it should work. 
This may be an issue to raise on the Daijisho Github. 

I'll see if I can find a workaround, however it seems like even before GammaOS, some people we having issues with this on other Android devices.

inteyes, 2023-07-09:
I've looked into this a bit more.

There is an option in daijisho, which transforms the path to the uri and provide output similar to mentioned above. For example, if I change {file.path} to {file.uri} within "Settings - Inspect players - dreamcast - io.recompiled.redream", then rom path looks like this in the log:

```
07-09 18:20:07.013  1367  1367 D daijishou_debug: AmStart args: [-n, io.recompiled.redream/.MainActivity, -a, android.intent.action.VIEW, -d, content://com.android.externalstorage.documents/tree/74E9-F089%3AGames%2Fdreamcast/document/74E9-F089%3AGames%2Fdreamcast%2FJet%20Grind%20Radio.cdi]
07-09 18:20:07.014  1093 10767 I ActivityTaskManager: START u0 {act=android.intent.action.VIEW dat=content://com.android.externalstorage.documents/tree/74E9-F089:Games/dreamcast/document/74E9-F089:Games/dreamcast/Jet Grind Radio.cdi flg=0x10000001 cmp=io.recompiled.redream/.MainActivity} from uid 10119
07-09 18:20:07.238  7616 11378 I redream : opening /tree/74E9-F089:Games/dreamcast/document/74E9-F089:Games/dreamcast/Jet Grind Radio.cdi
07-09 18:20:07.239  7616 11378 W redream : failed to open /tree/74E9-F089:Games/dreamcast/document/74E9-F089:Games/dreamcast/Jet Grind Radio.cdi
```

Still have no luck, it's now just "failed to open" :) But yeah, it's not a GammaOS issue after all, feel free to close this issue. Thank you for your help.

PS: just found out that Mupen64Plus FZ Pro executing games via frontend with no problems using {file.uri} macro, despite having the same file access permissions as the others :)

TheGammaSqueeze, 2023-07-10:
So to confirm, it looks like the latest version of Redream has this issue. 

I've attached an earlier build that works, you just need to enable access to all files permission. Newer versions seem to of have regressed this option for some reason. One to take up with the Redream dev.

Redream 1.1.98:
[redream_1.1.98.zip](https://github.com/TheGammaSqueeze/GammaOS/files/12001984/redream_1.1.98.zip)

Video demonstration:
https://github.com/TheGammaSqueeze/GammaOS/assets/116582950/62a08aef-d7d2-44cb-b474-d7810fcd6cee

inteyes, 2023-07-10:
Just tried it, now it works like a charm!

I guess [Google does not allow publishing](https://support.google.com/googleplay/android-developer/answer/10467955?hl=en#zippy=%2Cpermitted-uses-of-the-all-files-access-permission) applications in the market if they have broad file access permissions without compelling reason, so maybe that's why this options was removed.

I will close this issue since it's not GammaOS related (at all) as we found out. Thanks for your help and sorry for bothering :)

Cheers!