TheGammaSqueeze/GammaOS #29: Issues with the RG405M GammaOS

CruxisMithos, 2023-07-10:
Hello All, 

I just installed GammaOS on my RG405M following the RetroGameCorps video, and I have been noticing a couple of issues.

-  My unit will have the screen off, but not turn back on when pressing the power button and will randomly turn back on when pressing Home+power+Vol up (some times nothing happens when holding the buttons to varying amounts of time. 
- Some times when I get the enter PIN screen it will not recognize my PIN unless it turns back on after a reset. 

In short, my main issue is that my screen will sometimes refuse to respond/turn on when pressing power (or the combination I just mentioned). And it will sometimes turn back on when pressing it. 

Here are some details on my unit (in case they are helpful)

- I installed GammaOS from the most upto date Anbernic Stock OS...I think it was 1.18? 
- I removed my SD card during the install but put back in after the install was complete (do I need to wipe the SD card as well?)
- I have it setup with gesture navigation and set it to not require PIN if I am in my house (my way of avoiding the need to enter a PIN). 

I enjoyed the reduction of sharpness and am eager to use the unit more but I can't tell how to consistently turn on the unit/screen.

TheGammaSqueeze, 2023-07-10:
Hi, just out of curiousity, can you try disabling gesture navigation? 

There may be an issue with this on LineageOS which GammaOS is based on, only affects some users. You could also try toggling the performance mode to normal to see if that helps.

I would also suggest a reflash of the firmware and a factory reset, follow the instructions here, also make sure to run the EraseUserData.bat script in your case:

https://github.com/TheGammaSqueeze/GammaOS#instructions---upgrading-to-v13-from-gammaos--stock-os-with-unlocked-bootloader

CruxisMithos, 2023-07-10:
I will try moving away from gesture, but my performance mode had been on Default until about 30mins ago when I was able to boot it up while plugged in. I also ran the erase data file during initial setup…I just didn’t reformat my sd card.

closed 2023-07-15 by TheGammaSqueeze