TheGammaSqueeze/GammaOS #30: Issue with Wifi connection

karaschen, 2023-07-13:
My RG405M did not have any network connection problem with the stock OS/FW.
After I flashed GAMMA OS, when i connect to my home wifi, the OS says the wifi cannot provide network connection. I tried both 2.4G and 5G wifi signal and hotspot from my mobile phone, the console did not have network connection.  
I re-flash with both standard and lite version of GAMMA OS, the issue could not be solved.
Not sure if I missed any configuration in this OS or it's a bug.

TheGammaSqueeze, 2023-07-13:
Hi there, you may need to disable the adblocker, this is done by disabling private DNS, can you try this?
https://support.google.com/android/answer/9654714?hl=en#zippy=%2Ccontrol-data-use-with-metered-wi-fi%2Cfind-your-phones-mac-address%2Cprivate-dns

karaschen, 2023-07-14:
Thank you for your comment. But it seems not work for my system.

TheGammaSqueeze, 2023-07-15:
Then I am unsure of what the problem is, perhaps you should set a static IP and check again is private DNS is definitely disabled.

If that doesn't work, then I can suggest trying to go back to the stock firmware and seeing if the issue still persists.

karaschen, 2023-07-16:
It's wierd ,One of my friend also met this issue after installing this OS...

closed 2023-07-16 by karaschen