TheGammaSqueeze/GammaOS #32: Battery drain after installing gamma os on RG405M

arongkatz, 2023-07-13:
Hello

First of all thank you for your work doing this great os, i know this might not be the space to comment this but wanted to say it anyways, youre making a cool gadget a lot better fun gadget!.

Now, since i installed the os, ive been getting huge battery drains, first when i played some old systems like nes or gba the battery percent would go from 90% to 80% in a matter of minutes not even an hour playing. I changed the power profile to Power saving mode and still getting this battery life

I managed to charged it full 3 or 4 days ago and today it was at 1%. Could i be missing something? do i need to adjust some settings?

TheGammaSqueeze, 2023-07-13:
Hi, this might be down to calibration of the battery, the calibration data is lost when the bootloader is unlocked.

You'll need to let it drain completely, to the point of where the screen stops turning on and doesn't boot. 
From there, plug in the charger, turn the device on, then let it charge fully to 100% for a few hours. Let me know if this helps, thanks.

arongkatz, 2023-07-13:
> Hi, this might be down to calibration of the battery, the calibration data is lost when the bootloader is unlocked.
>
> You'll need to let it drain completely, to the point of where the screen stops [...]

Oh thanks, let me do that and come back to you with results

ShalokShalom, 2023-07-16:
Did this help?

ShalokShalom, 2023-07-16:
Did this help?

lucasgomeshora222, 2023-07-16:
I have the same question

inteyes, 2023-07-16:
This happened to me after unlocking the device. This method definitely helped restore the normal battery charge cycle. For me at least.

TheGammaSqueeze, 2023-07-17:
Closing as resolved for now

arongkatz, 2023-07-18:
So today i tried Max performance mode with a 100% battery. Played for maybe 5 minutes Starfox Assault on gamecube and now im on 73% battery life left. Is this expected?

arongkatz, 2023-07-18:
Thanks will get to 0% and unplug the battery and come back with results

closed 2023-07-24 by TheGammaSqueeze