TheGammaSqueeze/GammaOS #33: launch of Retroarch games failing >80% of time from daijisho

tinker-42, 2023-07-14:
Hello, just an fyi I loaded the most recent version of the gamma OS on my Anbernic 405m. My games are on a very fast SD card which is set up as portable storage. I set up multiple cores in Retroarch, and the problem that I’m about to describe happens with all of them regardless of rom size or core. When I set up the daijisho front end, I was careful to select the appropriate rom Directory and also made sure to use the 64 bit retro arch entry for a particular emulator core.   However, when launching a game from Daijisho it will often take several tries to launch where most just return to daijisho or in many cases Retroarch hangs with a black screen. If I launch the game from Retroarch directly, there is no issue, Everything works fine. If I look in the daijisho log after trying to launch there are no error messages, only the command to invoke the game. I don’t know if this is a scoped storage response or throughput issue, or some other factor. I was, however, thinking that it may be worth publishing a version of the gamma os without daijisho and using a different app launcher. I was wondering if you had any suggestions?

TheGammaSqueeze, 2023-07-15:
Hi, I've not had any issues personally with using an SD card as portable storage with Daijisho and Retroarch. And you're the first to report this kind of issue.

Do you have another SD card to try with to rule anything hardware based out?

You can try any launcher you want my installing one from the Play Store, or changing the default on in Android Settings > Apps > Defaults.

arongkatz, 2023-07-15:
Im also having this issue but just with Saturn games launching them to yaba sanshiro. It shows the splash screen and then comes back to game list on Yaba

ale296, 2023-07-15:
Same here, everything works except yaba sanshiro 2 pro from Daijisho.

TheGammaSqueeze, 2023-07-15:
yaba sanshiro 2 is known to have issues with newer android build due to scoped storage, I have yet to find a fix for this.

tinker-42, 2023-07-15:
I don’t think it’s related to the SD card.  I tried a couple SD cards and it’s the same with either. Both are fast and rated for android a Samsung pro plus and Evo select. I should’ve been more clear that the issue is Retroarch specific. I can launch standalone emulators like duck station or ppsspp gold or flycast. Looking at the logger they will all throw a malformed URL exception error but the game will actually launch. With Retroarch there is no error, Retroarch just hangs on launch. I tried clearing the android cache for Retroarch, I also tried turning off the global Shaders that are part of the base build, but no change. I have noticed with a bit of googling that people talk about similar issues with Retroarch hanging with Daijisho or launchbox on a number of different android devices. Jelos on my 353m work’s totally fine with similar sd cards and directory structure on the sd card. I will try disabling daijisho since it is not providing any benefit. I was mostly wondering if I turned it off is trebuchet there? I will also download nova. I will try to follow up with the daijisho developer.

tinker-42, 2023-07-15:
I changed the home app launcher to trebuchet. I am happy now, not sure what’s up with daijisho

closed 2023-07-17 by TheGammaSqueeze