TheGammaSqueeze/GammaOS #35: Navigation bars showing on some standalone emulators

arongkatz, 2023-07-17:
This may be more of a tech question than an issue but this didnt happen on stock os so i didnt know how to configure it. Im basically playing on yaba sanshiro and the nav bars just wont go away. Is there a way to remove them or do i need to go full tv mode?

dfreivald, 2023-07-17:
Had the same issue. It's hiding somewhere in the settings. It should be in both the pro/non-pro versions.

TheGammaSqueeze, 2023-07-19:
Is there an option in Settings > Graphics for Hide Status bar and Navigation bar?

It is mentioned here in the developers site:
https://www.uoyabause.org/static_pages/guide

[image]

arongkatz, 2023-07-21:
I disabled the navigation hint option but the line is still appearing on all apps and retroarch. I read somewhere this might be a bug on lineage?

dfreivald, 2023-07-25:
I used the "Hide Status bar and Navigation bar" setting and that worked for me.

closed 2023-08-01 by TheGammaSqueeze