TheGammaSqueeze/GammaOS #37: Can't Download

junkaddr, 2023-07-19:
Hi, your google drive link just errors out as there was too much interest. You can't actually download this.

Can you at least host a torrent?

eduardopatricio, 2023-07-19:
I just checked here and the download links work for me. 

Sometimes Google Drive gives out this message for other reasons, even if no one download the file. 
Are you logged in to your Google account on the browser? If not, logging sometimes "fixes" the issue.

Also, I don't understand the "at least"part of the question. The developer is doing a lot already, and out of the goodness of their heart.

junkaddr, 2023-07-19:
A torrent is a way for others to download the file without having to pray that Google doesn't block it for costing them too much money. It isn't an unreasonable request to be able to download the project. It doesn't matter how much effort anyone puts into it if it isn't available.

Thanks for your sanctimony though, really helpful.

poc-sm, 2023-07-19:
Same issue here when I try to download the file (lite or not) / don't have google account
For next release I will make a torrent file or hosted a mirror elsewhere

TheGammaSqueeze, 2023-07-19:
Hi all, for the next release I will look to host the files on Gitlab releases itself, at the moment it's not possible due to the file size limit of 2GB per file.

At the moment in time, I am travelling for work so don't have somewhere that I can keep this seeded as a torrent. I'll see if I can upload elsewhere on a temporary basis though. A workaround is to sign in with a Google account first, it may let you download after this.

closed 2023-07-24 by junkaddr