TheGammaSqueeze/GammaOS #41: Feature Request: Performance Mode Switcher and Button Layout Switcher

anup-khanal, 2023-07-21:
XBOX and NS button layout switcher button and Performance mode switcher button on the quick settings bar when you swipe down would be nice to have.

arongkatz, 2023-07-23:
This!

dfreivald, 2023-08-02:
this is for future Gamma: https://developer.android.com/develop/ui/views/quicksettings-tiles

jl-casella, 2023-10-27:
Hi guys, first of all, thanks for the GammaOS. Having said that, as a user and dev I also faced this issue, so I worked on an app to improve this experience.

I am releasing it today. So I was first letting you know here, before posting on some other forums.

You could take a look at it on Google Play. It already ships Button Layout Switcher in tiles. I will integrate soon the one for perf modes too. You can also add a Tile to quickly access the app, where you can tweak a lot of other device stuff.

The inspiration is a mix of CPU-Z. and the Settings app from the new AYA Neo Pocket AIR. For the Tiles, my next goal is to copy as much as possible the UI of the ones from Odin 2.

Once installed, the app will ask you for some permissions. Once you grant them all, disable notifications from the Magisk app, so you do not get a notification each time it uses `su` access:

[image]

Google Play link: https://play.google.com/store/apps/details?id=com.controlcenter.android

closed 2024-02-23 by TheGammaSqueeze