TheGammaSqueeze/GammaOS #43: Unable to connect to any device using MiraCast/ScreenCast

kaleben0, 2023-07-21:
Unable to connect to any device using MiraCast/ScreenCast for RG405M

- This worked on stock OS prior to GammaOS
- Unable to connect to any wireless display (Tried Chromecast, Roku, Firestick, and Windows 11 Wireless Display)
- No error message, just fails to connect

[image]

Not sure what the issue is, but any help is very much appreciated

eduardopatricio, 2023-07-21:
Did a quick test on my end here now to see if it's a general issue.
I could successfully cast it to my Nvidia Shield TV.

E.

kaleben0, 2023-07-21:
Ok, I think I found the problem and It is not an issue with GammaOS. 
Apparently a lot of Miracast devices don't like the "auto-accept" setting for when a device tries to connect to it. 
I switched my Windows 11 (Wireless Display) Roku & Firestick to prompt when connecting and it worked instantly. 
Still can't get the Chromecast to work, but it is an older one so that might be the issue. 

Thanks for checking

closed 2023-07-21 by kaleben0