TheGammaSqueeze/GammaOS #45: Unable to turn on after leaving for a while (rg505)

tombroskipc, 2023-07-22:
My RG505 cannot open if I leave it for a while (like 1 hour).
When I charge it. It auto turns on back when it hit 100% battery.
Love your work. But can you do anything about it?

TheGammaSqueeze, 2023-07-24:
Hi, I can only suggest flashing the firmware again, and don't install any additional apps and see if the issue reoccurs. Don't enable navigation gestures and don't enable screen lock.

Follow this guide/video to reflash, ensure you also run the EraseUserData script again too.
https://github.com/TheGammaSqueeze/GammaOS#instructions---upgrading-to-v13-from-gammaos--stock-os-with-unlocked-bootloader

poc-sm, 2023-07-25:
Same issue last week for me but didn't investigate (happen only one time)
Navigation gesture is disable but I've got a screen lock and always on max perf

tombroskipc, 2023-07-25:
I didn't enable screen lock and Navigation gesture.
Let me reflash the firmware if the issue still there.

closed 2023-08-01 by TheGammaSqueeze