TheGammaSqueeze/GammaOS #47: USB / Bluetooth Audio Routing

kaleben0, 2023-07-23:
I am not sure if this is an limitation of the hardware or something else.

Trying to use either a USB audio device (USB-C Headphones / Small USB Speaker) or a Bluetooth audio device (Earbuds) and neither is working

It appears that the devices are noticed by the OS, but the audio will only route to the internal speaker or the 3.5mm port. 

I checked the developer options and automatic audio routing is still enabled. 

There is even an option to select the earbuds when using something like Spotify, but it still goes to the speaker. 

This is not a deal-breaker, just me testing the limits of the RG405M and Gamma OS. 

Any help is appreciated

mugenoesis, 2023-07-24:
More or less the same thing happening to me. 

I've tried with my bluetooth headphones though and the audio worked fine. 

I have tested 2 different amp/dacs: the THX onyx lights up alright but audio continues playing over the speakers, the other one is a cayin ru7 and that doesn,t turn on at all, possibly not enough power.

 I tried with a pair of usb iems specifically the moondrop dsp quarks, this does connect and again the audio keeps playing over the speakers. 

I know with both the thx onyx and the quarks they connect because usb audio player recognizes that they are connected.

TheGammaSqueeze, 2023-07-24:
Hi, thanks for this info. I'm not able to replicate the issue with the usb-c and bluetooth headphones I have at home so this is a bit concerning.

I take it that this worked on the stock os fine?

kaleben0, 2023-07-24:
Honestly I went straight to GammaOS a day or two after I got it. 

Didn't tinker with it much before changing the OS.

mugenoesis, 2023-07-24:
I just got my 405m today and straight away switched to gamma so dunno if it works on stock. I just tried out another dac (the one that comes in the box with the rg nano) and my audeze maxwell over usb. Both connect and both and do not get audio routed to them. Puilling the dac out while music is playing does pause the music, so some of it is working. Just not the audio bit :)

kaleben0, 2023-07-24:
> Do you have the name and model of the headphones you are trying?

I have tried these so far

https://www.amazon.com/gp/product/B07S5JVP78/
https://www.amazon.com/gp/product/B07MX1XL1V/
https://www.amazon.com/gp/product/B09K3S9VRL/

kaleben0, 2023-07-25:
I think I found the issue with the Bluetooth. 

It appears to be a bug with with Lineage OS.

https://gitlab.com/LineageOS/issues/android/-/issues/2812

I tried this and not only did it work, but bunch of other Bluetooth options showed up in the developer options. 

It had no effect on USB audio routing, that still does not work.

mugenoesis, 2023-08-24:
So havn't had time to do anything with this sytem for awhile but i finally reset it back to stock firmware. I can confirm that usb audio routing works as expected on stock with the devices i mentioned before except for the cayin ru7 (i still wonder is it a power thing)

fonkfader, 2023-12-26:
i think my issue is related to this, M8 headless app (https://github.com/v3rm0n/m8c-android) have distorded sound and unresponsive using a tennsy 4.1 board plugged in usb-c otg. i tried to use bluetooth headset but the app cannot see it (you can select audio output in the app), outside this app i've tried 2 usb dac and the audio keeps playing over the speakers / original audio out port not the usb dac, same issue using bluethooth headset, i see the BT headset connected can change volume etc... but the sound never go to it only keeps playing over the speakers... i have read some comment somewhere that the m8c app work fine on original firmware (but i prefer keep gammaos build for the emulation part...)