TheGammaSqueeze/GammaOS #51: RG505 wakes up if buttons pressed shortly after power button puts to sleep

Luk37astic, 2023-07-25:
After pressing the power button to put to sleep, the sooner other buttons are pressed, the more immediately it wakes back up.

Also happens randomly even after a long time of being asleep. (Spamming buttons will wake up the device)

MetroidFusion, 2023-07-26:
I can confirm this issue on the RG505. The device will wake up if buttons are pressed - as mentioned, it also happens even after being a sleep for a while. My device had it's battery drained significantly over night due to this issue, since my protective case apparently pressed one of the buttons while not in use.

/M

TheGammaSqueeze, 2023-08-01:
This is expected behaviour as the controller is emulating a Xbox Series S gamepad via Bluetooth. Bluetooth gamepads wake the device on button presses.

I have implemented a workaround which checks if the screen is off, this happens every 5 seconds or so. If the screen is off, then the buttons inputs will be blocked. 

If for any reason that the screen is turned on, for example via notifications, this can make the buttons register again, thus keeping the device awake.

Unfortunately I don't have a workaround for this except for turning notifications waking the screen option off.

Closing this issue, hope this explains the behaviour.

closed 2023-08-01 by TheGammaSqueeze