TheGammaSqueeze/GammaOS #54: RG405M: Vibration motors not recognised in standalone emulators.

jkmoogle, 2023-07-31:
Some standalone emulators that work properly on stock OS are not detecting vibration support in the device. Confirmed myself that Drastic, Duckstation, AetherSX2, and Dolphin (Play Store) have the same issue. There may be more, and this may also be an issue with the RG505 as noted by Reddit users.

poc-sm, 2023-08-01:
Hello 
I just test on 405M and working well with Dolphin 
Didn't test other, seems related to 505 only

TheGammaSqueeze, 2023-08-01:
Hi, for AetherSX2 and Duckstation, you should enable vibration via the touchscreen options in those apps.

Dolphine MMJR also works with vibration.

This is not a GammaOS issue, rather the driver that Anbernic provides does not support controller rumble. Instead they have preconfigured the emulators with what I've mentioned above.

closed 2023-08-01 by TheGammaSqueeze