TheGammaSqueeze/GammaOS #55: RG505 randomly restarts or turn off when idle

combo123, 2023-08-01:
The RG505 randomly restarts or completely turns off when the device is in idle. If it turns off, it flashes the Gamma Os logo and then completely shuts off, after that it is impossible to turn it back on (no reaction from any buttons, does not even charge), the only way is to phisically open the back and remove/reinstall the battery.

TheGammaSqueeze, 2023-08-01:
Hi, you can force a reboot without disconnecting the battery by holding down volume and power button.

If the issue persists, I recommend reinstalling the stock firmware again, then install GammaOS again.

combo123, 2023-08-02:
> 

Yes, also not reacting to that combination. Will try your second recomendation this week, will report back how it goes, thank you

combo123, 2023-08-02:
> Hi, you can force a reboot without disconnecting the battery by holding down volume and power button.
> 
> If the issue persists, I recommend reinstalling the stock firmware again, then install GammaOS again.

Small update, I left the device to charge overnight (while it was turned off). So far the problem has not came back, has not turned off once, if anything new I will report back again :)

AGvin, 2023-08-05:
Hi here!
Looks faced with same issue but for RG405M
On idle or sleep system restarted.

Already tried to fix this with restoring stock OS, and than again install GammaOS, but issue still reproduced.
It looks caused with some trim onTrimMemory service

[image]

combo123, 2023-08-09:
> Hi here! Looks faced with same issue but for RG405M On idle or sleep system restarted.
> 
> Already tried to fix this with restoring stock OS, and than again install GammaOS, but issue still reproduced. It looks caused with some trim onTrimMemory service
> 
> [image]

Could you try turning it off and leaving it overnight to charge? That actually worked for me, so far so good

AGvin, 2023-08-10:
> Could you try turning it off and leaving it overnight to charge? That actually worked for me, so far so  

Tried to do that, but it doesn't help

BadgerB01, 2023-08-22:
I can also confirm, brand new 405m, first install of gamma OS 1.3.3 created issue of device resetting after screen timeout, the power button/device becomes unresponsive, soft reset needed to turn back on, stock os recovered yet problem continued even on "stock", reinstalled Gamma and problem continues. Have set time out to longest duration, this prolongs the reset/reboot cycling, and means you can use the power switch to put in sleep successfully as long as turned back on in the original 10 minutes time out window. Overall problem continues and is very frustrating

L-K-M, 2023-08-29:
I have a very similar issue. I just installed 1.3.3 on a new RG505. When I put it to sleep by hitting the power button, after a while (it seems to be anywhere between 2 and 30 minutes, but I'm not entirely sure), it restarts on its own. It's fully charged.

SlowPixel, 2023-10-04:
Yes, the problem is the same on 505. I noticed that this often happens when applications are open in the background and if they are not closed and put to sleep, a shutdown may occur.
Perhaps this coincides, but I have not yet identified the exact reason.
1.3.3 Last
And I put the wasps on a clean drain.
I really hope the new build will be more stable.

closed 2024-02-23 by TheGammaSqueeze

Kyo-Tachibana, 2024-05-17:
> I can also confirm, brand new 405m, first install of gamma OS 1.3.3 created issue of device resetting after screen timeout, the power button/device becomes unresponsive, soft reset needed to turn [...]

I have the same problem, I installed the latest version full of gamma OS and when I press power to enter in sleep mode then turn off, or if it enter in standby turn it's turn off to...

kkakele, 2024-06-12:
Does anybody found a fix for this? Everytime i push sleep button the rg505 crash and have to force restart.... Sorry for my english.Hope to find a solution soon....