TheGammaSqueeze/GammaOS #56: Feature Request: profiles for common emulators available in the Play Store

JamesRHarris, 2023-08-02:
I recently installed Dolphin from the play store. While it worked I suspect performance can be greatly improved by a good bit of tuning. Instead of leaving it to each user to tune the emulators; It would be great if the OS either included a tuned configuration for common platforms. Alternatively an integrated app to pull in such tuned configuration files would also be a good option.

poc-sm, 2023-08-08:
There are a spreadsheet about game compatibility 
https://docs.google.com/spreadsheets/d/1Vf7SIS7ecWa_J301h0mb2bxMWBMtKLvcpAFyaa5-LDc/edit#gid=1908749862
Seems there are no good choice for every game on gamecube settings
Also check this out may be it could be helpfull https://www.youtube.com/watch?v=D0dGx-vTCpI

TheGammaSqueeze, 2024-04-04:
This will not be implemented due to maintenance overhead.

closed 2024-04-04 by TheGammaSqueeze