TheGammaSqueeze/GammaOS #67: Fixing bricked state: getting error “Failed: [PS2257]Uart send error” from spdflashtool

jsoryu, 2023-08-27:
Hello Gamma. First of all, thanks for creating this awesome firmware.

I’ve unfortunately had issues where device won’t turn on. In some occasions, go into boot loop.

Fact that spdflashtool detects the device, I assume it’s in bricked state and trying to reflash the original firmware.

I’ve been getting message “

TheGammaSqueeze, 2023-08-27:
You may need to hold down the home/back button, Power button and Vol down button to force it into a flashing state for the SPD tools to allow flashing.

I assume you have installed all the correct drivera and are on a Windows machine?

jsoryu, 2023-08-28:
Answers are yes on all of those you’ve asked. Process begins after I’ve pressed the mentioned buttons (I’ve watched the YouTube video included in the instruction, as well as one recently mentioned in Anbernic discord.

I must’ve downloaded and tried virtually all variations of SPD tools; luckily they came with own drivers. (URL to download were included in all variations)

Regardless, Anbernic support just got back to me that they’ll send new board to me. Still, it’s a shame that I still can’t get past this.

I suppose we can close this ticket, but welcome on any suggestions and guidance.

Thanks

closed 2023-11-22 by TheGammaSqueeze

nyckenn, 2024-01-15:
I have the same problem....have changed many cables and laptops but not working. Could you tell me the solution?

RandomTurtle, 2025-01-06:
Same problem here :(