TheGammaSqueeze/GammaOS #70: Wifi randomly disconnects - rg405m

Beltaidan, 2023-08-30:
Hello,

WiFi randomly disconnects and will not auto-reconnect unless i disable wifi and force it to reconnect to my network. It happens very randomly but is annoying bc there is no way to see i am connected while playing games without exiting. if this happens while playing i can no longer earn retro-achievements for whatever game im playing.

dfreivald, 2023-09-14:
This happens on stock as well. I don't know that there is a better wifi driver available for this chipset.

DjFIL007, 2023-09-29:
Wanted to add related to this. I have to fully reboot my device occasionally when Wifi doesn't want to reconnect. When trying to access the Wifi settings menu the menu will go 'not responding' and OS will recommend I close or restart it. After rebooting the wifi adapter usually shows as off/disabled, and but responds quickly to turning it back on and detecting my network(s).

Have had this happen 2-3 times, usually as I leave my device on almost all the time (in sleep)... going between 2-3 different Wifi networks I have it paired with.

closed 2024-02-23 by TheGammaSqueeze