TheGammaSqueeze/GammaOS #71: Unable to flash vbmeta_a on RG505

geoffgs, 2023-09-02:
Started with a stock RG505

Worked around the Linux issue with unlocking the bootloader.
Started to flash from Ubuntu 22.04, but it failed at a few points.
Switched to a Windows 11 host to unbrick, back to stock.
Tried again from Windows with the batch script, but flashing of vbmeta_a didn't complete even after 20 minutes
Unbricked back to stock again
Switched back to Linux, started with flashing vbmeta_a again manually, but again it doesn't complete after 30 minutes
`fastboot flash vbmeta_a vbmeta_custom.img 
Sending 'vbmeta_a' (1024 KB)                       OKAY [  0.026s]
Writing 'vbmeta_a'    `

I'm using the USB cable included from Anbernic, and it has no trouble with the debrick tool so I'm fairly confident it's fine. I've flashed Lineage plenty of times on a number of devices, and this is unusual to me. Any idea where I'm having trouble? Any help much appreciated.

closed 2023-09-15 by geoffgs

Ufoex, 2023-11-02:
I see that you closed it, I have the same problem as you, could you give me the solution you found?

geoffgs, 2023-11-05:
> I see that you closed it, I have the same problem as you, could you give me the solution you found?

There was no perfect fix, but: 
1. Restore to stock with the Windows-only utility
2. Make sure you're in [fastbootd](https://source.android.com/docs/core/architecture/bootloader/fastbootd#adb-commands) before flashing: `adb reboot fastboot`
4. Repeat