TheGammaSqueeze/GammaOS #73: Bootloop

Bob021021, 2023-09-06:
Hello,

Great work and a great project you are working on here.
I received my RG505 this week and wanted to start optimizing right away.
Following the instructions here and the video, I worked my way through the process.
After the installation is complete, I should press the power button and wait for GammaOS to start.
Unfortunately, my device has been in bootloop mode since then (similar to issues # 14 with the same image on the device).
It is only the following on the device
INFO: LOCK FLAG IS: UNLOCK!!
Warning: Lock Flag is: Unlock, Skip Verify!!!

The instructions were followed and no SD card was used during the process.
Unfortunately I can't get any further to solve this problem.

What has been tried so far:
- waited 24 hours for the battery to die - until then a permanent bootloop
- An attempt was made to connect the device to the PC (WIN 10) but the device is no longer recognized here
- only in the brief moment of switching on after shutting down due to an empty battery (before the boot loop begins) does WIN recognize a connected device - for about a second.
- Various key combinations on the device were tried to escape from this mode, but without success. Probably also because I don't know the right combination for this either.
power does not work
power and Vol+ does not work
powre and vol- seem to cause a reboot, but the bootloop begins again

I hope someone can help me with a workaround here.

Greetings
Bob

UrDone4, 2023-09-07:
Have you tried the steps in the instructions under "Help! My device is no longer booting! Or I want to go back to Stock OS!"

Here is a link to it: 
https://github.com/TheGammaSqueeze/GammaOS#help-my-device-is-no-longer-booting-or-i-want-to-go-back-to-stock-os

Bob021021, 2023-09-08:
Hello,

Thanks for the help.
I immediately tried to do it according to the instructions.
The process is also carried out, but the RG505 then restarts with the black screen and the text:
Info: Lock flag is: unlock!!
Warning: Lock Flag is Unlock, Skip Verification!!!
and remains in bootloop mode

I've tried several times now and waited a long time to see if the screen showed something else, but it still remains in bootloop mode.

I'll try it a few more times this weekend.

If there are any further hints, I would be happy to try them.

Bob021021, 2023-09-08:
[image]

UrDone4, 2023-09-08:
The text you are seeing is expected on the boot screen. I am unsure why the device would be bootlooping, however.

dfreivald, 2023-09-14:
I was bootlooping as well - but reflashing stock fixed it, and I was able to re-install without issue.

Bob021021, 2023-09-15:
Hello,

@dfreivald
Thanks for your information.
Have you done anything else to flash the standard files besides the tool, which I also attached as a screenshot?

@UrDone4
@ all
I tried several times this past weekend. About 30 times. Once I also saw the former start screen of Anbernic, but here too it restarted every 15-30 seconds.
I'll try it this weekend on another PC with a different version of Windows.
I will report how and whether things are progressing.

Greetings
Bob

dfreivald, 2023-09-17:
I used the same spdflash tool using the unbrick instructions.

closed 2023-11-22 by TheGammaSqueeze