TheGammaSqueeze/GammaOS #74: Make this project FOSS for futureproofing?

d3adb5, 2023-09-08:
Though this repository is on GitHub and we can easily submit pull requests, the source code for GammaOS itself isn't included here, only the scripts necessary to flash the files included in the ZIP archives linked in each release.

I'm not an Android developer, so I can't promise I'll be able to contribute much to the project overall, but I'm wiling to learn my way around Android development. There are definitely developers out there who can and are willing to contribute to GammaOS. One of the advantages of making the project free and/or open source is that in the future, when the original developers have already moved on from it, other people are still capable of working from where they left it.

This is a selfish request, I admit, and if the source code is already publicly available elsewhere, please disregard this issue.

poc-sm, 2023-09-08:
hello ! 
Check this issue #7

d3adb5, 2023-09-09:
So if I understand it correctly, GammaOS is a simple modification (I'm assuming in terms of system apps and other configuration) of the LineageOS GSIs? Again, not an Android developer, but is it at all worth forking the LineageOS source and working on top of it with some install scripts and whatnot? Basing myself off of [this comment.](https://github.com/TheGammaSqueeze/GammaOS/issues/7#issuecomment-1584670981)

Could @TheGammaSqueeze help clarify this?