TheGammaSqueeze/GammaOS #76: Issue saving GBA games when using physical button

sgano, 2023-09-15:
There seems to be a bug when using GammaOS and GBA where the games don't save if you use the physical home button (under the d-pad on the left side of the RG405M) to exit games.   However, if you press L3+R3 and manually close the game the saves all function properly.     For reference also see:  https://www.reddit.com/r/ANBERNIC/comments/1598fwk/problem_with_save_games_on_gamma_os_in_rg405m/

dfreivald, 2023-09-30:
Weird. Stock OS and my phone do the same...

dfreivald, 2023-09-30:
OH... looks like you can bind the back button in RA as of GammaOS 1.2

sgano, 2023-10-12:
The interesting thing is the other systems seem to save properly with RA using the home/back button; just GBA seems to have this saving issue.  The pre-bound back button in GammOS works fine for the rest of them. I guess it could be an issue with that particular system within RA.  Though the way you describe the state changes does make a lot of sense as to how the data is lost; it is just puzzling with it isn't consistent across systems.

TheGammaSqueeze, 2023-11-22:
> Weird. Stock OS and my phone do the same...
> 
> edit: confirmed with others the RP3+ and Odin Lite also does this, and that's how android and RA works with save states. Long pressing home just puts RA into the background (like alt-tab in windows), without ever telling it to exit or save state. Next time you launch the game from Daijisho, Daijisho automatically terminates RA that was running in the background - and poof, your progress is gone.
> 
> Interesting thing is that Gamma has experimented with intercepting the home button, and the changing the resulting keypress only when RA is in the foreground. This allowed hijacking home and swapping it to another key that can be bound to save/quit in RA only. I'm worried that will break other functionality in RA, so not sure if if this was or will be rolled into GammaOS. I'll let the man himself comment on this :]

I will indeed roll this in, it won't break any other functionality though and users will be able to remap this button to something else if desired. 

The normal RA behaviour in Android is that going to the home screen either via the physical home button or the on-screen one will keep RA running in the background. This doesn't actually save your progress in game though. Depeding on the core, saves may not be flushed to the disk until you actually close the game from within the RA menu, hence the save loss here.

sgano, 2023-11-27:
Thanks for looking into this and for all the hard work it is much appreciated!

TheGammaSqueeze, 2023-11-27:
If using the gpsp core, you can enforce that saves are flushed to disk every 10 seconds. Go to the core options, under the Backup Save Method, change this to libretro.

You will probably want to ensure you have a savestate with correbt progress before you do this. I'll make this the default for new installs going forward.

https://github.com/TheGammaSqueeze/GammaOS/assets/116582950/ef164966-b1b3-42d6-8453-0a0070a38228

TheGammaSqueeze, 2023-11-27:
Changing the gpSP core settings fixes this particular issue. Change Backup Save Type to libretro and setting saveram interval to 1 second eliminates this issue. I'll set this as the default going forward for new GammaOS installs in the next version. 

If you change this, be aware that you will lose your current save. 

Uploading Screen_Recording_20231127_053030_RetroArch Plus.mp4…

closed 2024-02-23 by TheGammaSqueeze