TheGammaSqueeze/GammaOS #79: Expand notification area width in landscape mode

d3adb5, 2023-09-22:
The RG405M (and I'm assuming the RG505 as well) screen is perpetually in landscape mode, likely by design as most Android games will default to operating in landscape. In the fourth beta version of Android 12, changes were made to the pull-down notification area when in landscape mode for the sake of more elongated aspect ratios: [notifications are now centered with huge margins on either side](https://www.androidheadlines.com/2021/08/android-12-odd-change-notifications-landscape.html) so that they're easier to read and reach.

Sadly when this applies to our dear RG405M, the margins don't scale down or get eliminated due to the screen's relatively low width in landscape mode, and so we get notifications that are rather crammed:

[image]

Due to apps preferring landscape mode over portrait, I feel like making Android recognize the different natural display orientation isn't the proper solution to this, but I have found no way to eliminate or lower those margins to an acceptable degree.

PhucHauDeveloper, 2024-02-28:
> The RG405M (and I'm assuming the RG505 as well) screen is perpetually in landscape mode, likely by design as most Android games will default to operating in landscape. In the fourth beta version of [...]

You can use the magisk module called "Better QS" version 1.0 by hani, which will help your QS panel no longer have excess space.

d3adb5, 2024-03-08:
> You can use the magisk module called "Better QS" version 1.0 by hani, which will help your QS panel no longer have excess space.

The problem is with the notifications not spanning from one side of the screen to the other, not the quick settings panels. I haven't been able to find this Better QS module, though. Do you have a link?

closed 2024-04-04 by TheGammaSqueeze