TheGammaSqueeze/GammaOS #82: PSP games in vertical orientation

PackingCenter, 2023-09-27:
Hi. Downloaded PSP core via Daijsho.  Psp games started in wrong display orientation - vertical, not horisontal. Gamma OS. How to fix?

poc-sm, 2023-09-27:
I recommand to use PPSSPP
But to switch from retroarch ppssp core  when you started a game press L3 + R3 and go to settings -> video -> video output -> video rotation and switch to 90 deg
Don't forget to save the settings for the core

dfreivald, 2023-09-30:
PSP was never configured in RA, because we assumed people would know to install PPSSPP before launching a PSP game from Daijisho. 

After you install PPSSPP, you have to edit the PSP platform in Daijisho (click on the pencil), and change it to launch using PPSSPP instead of retroarch 64.

closed 2023-11-22 by TheGammaSqueeze