TheGammaSqueeze/GammaOS #84: PS Remote play don't work in Anbernic 405m

andretiton, 2023-10-07:
Hi, when I try to run PS Remote play app on 405m with GammaOS I receive this msg: Something went wrong (88001003). How to resolve?

andretiton, 2023-10-07:
[image]

poc-sm, 2023-10-09:
Hello
What's app do you use ?
The official ps remote play ? Or Chiaki ? Or PsPlay ?
Can you tried the three ?
The error seems related to a rooted phone ou unlocked bootloader 
It seems you need to add a module in magisk 
https://www.youtube.com/watch?v=O60HIMaU-6Q

andretiton, 2023-10-09:
> Hello What's app do you use ? The official ps remote play ? Or Chiaki ? Or PsPlay ? Can you tried the three ? The error seems related to a rooted phone ou unlocked bootloader It seems you need to add a module in magisk https://www.youtube.com/watch?v=O60HIMaU-6Q

Hi, I try Chiaki and run ok. I lose anything run Chiaki and not oficial PS app?

andretiton, 2023-10-10:
> I don't think, don't worry [thestr4ng3r/chiaki#171](https://github.com/thestr4ng3r/chiaki/issues/171)

Thank you, I feel calmer.  But I also meant in terms of performance.  Any known issues?

closed 2023-11-22 by TheGammaSqueeze