TheGammaSqueeze/GammaOS #85: Device physical orientation isn't landscape

RomaRogov, 2023-10-12:
Hello! 
Gyro sensors behaviour is different from stock OS. At first I was sure that it's a good thing that behaviour is changed to smartphone-like but now I'm experiencing problems in a lot of games.
So, AIDA64 shows that X and Y gyro axes are swapped and works like device physical orientation is portrait, but actually it's not.
Can you please make sensors to work as they are on stock OS? I'm using it for aiming in some old shooters ports and it's pretty important thing IMO.

andiweli, 2023-10-23:
I too have 90 degree rotated text during boot, as well as the taskbar on the right hand side instead of the bottom.

Also the performance/standard mode settings aren't saved during a reboot.

mgar, 2023-11-02:
Not sure if this is related, but I noticed that when you apply a shader with scanlines, they appear vertical rather than horizontal.

TheGammaSqueeze, 2023-11-22:
Yes the 405 series has a portrait display, this makes sensors behave a bit differently. 405V is even worse in this regard.
I will need to investigate and fix this in Android.

RomaRogov, 2023-11-22:
Please tell if you need any help with that, I can spend some time to test, try and inspect different solutions. I have RG505 though

closed 2024-02-23 by TheGammaSqueeze