TheGammaSqueeze/GammaOS #89: WiFi connecting on first  launch

pedrosegadas, 2023-10-21:
Hey, 
I installed GammaOS, but on the first launch, right after selecting the language, when I need to connect to a WiFi network, it just doesnt locate any network... And Im stucked on that screen, cant proceed...
Anyone knows what can it be??
[image]

pedrosegadas, 2023-10-21:
PS: Now I restored the Stock OS, and still cant turn on the WiFi... Before I started the installation of Gamma OS I didnt have any issues with WiFi

TheGammaSqueeze, 2023-10-21:
This only happens if you try to flash the 405m firmware on a 405v.

Please refrain from doing this, as there are some differences between these devices.

In the meantime, use the following unbricker for the 405v:

Anbernic RG405V Unbricker

Did you mess around with your device and now it won't boot? Don't worry I've got you covered. 
This unbricking tool contains the stock RG405V firmware (v1.23) and will allow you to return back to factory settings.

Download: https://drive.google.com/file/d/1idODOwE197mRHwMHDU_Y7bnAiWiackbN/view?usp=sharing

Instructions (same as RG405M, Windows only): https://github.com/TheGammaSqueeze/GammaOS#help-my-device-is-no-longer-booting-or-i-want-to-go-back-to-stock-os

pedrosegadas, 2023-10-22:
Got it... Last question, The message "Lock Flag is unlocked" at reeboot can be removed?
@TheGammaSqueeze 
Thanks!!

wansadi, 2023-10-27:
I may have just run into the same issue. In my case, I was able to connect to a wifi, but it keeps telling me that there's no internet connection for the connected wifi. So there is no official firmware release for 405v yet?

androidan, 2023-11-05:
I have flashed the 405m on my 405m correctly and have same issue, how can I solve it? thanks

Bellathepug, 2023-11-18:
What does “lock flag is : unlock!!!” mean? My 405v doesn’t unlock..

TheGammaSqueeze, 2023-11-22:
lock flag is : unlock!!! means you are already unlocked. No need to unlock again.

closed 2023-11-22 by TheGammaSqueeze

tbobruns, 2024-01-31:
I ran into this same issue using RG405M.GammaOS_v1.5.1.7z on my RG405M. What could be causing this issue? Wifi refuses to connect.

hpcraftwerk, 2024-04-26:
I am having this same issue, flashed with Gamma 405v and Gamma 405v Lite, neither one will see Wifi.

Used the Unbricker with Firmware 1.23.pac to restore and try each time.

Mouer78, 2024-05-07:
Same exact problem as those above. 
I have a rg405v.
Made sure to us the Gamma 405v firmware. 
Will not see WiFi afterwards. 
I also tried Gamma 405v lite, same problem. 
I used the Unbricker tool and got the WiFi back in the stock OS. 
I went back and forth with this process about 4 times before finally giving up.

tomfuri, 2024-06-25:
someone can help please ?

TheGammaSqueeze, 2024-06-25:
Anbernic keep insisting that nothing has changed from their end.

Doubt

I've worn them down into sending a new 405v from factory, once I have this, I'll be able to see what the problem is for these new units.

This will be a couple of weeks

strike76, 2024-06-26:
@TheGammaSqueeze Same problem for me, RG405V received on 24/06/2024
Do I need to reset the stock firmware or is there a way to get the wifi to work?