TheGammaSqueeze/GammaOS #92: Do we have an ETA for Gamma 405v release version?

wansadi, 2023-11-03:
I somehow flashed so called beta version in to my 405v. For the most part it works pretty good except for fan not working and no internet connection when connected to any wifi. Are we planning for a 405v version yet?

rogiervanroon, 2023-11-08:
I'm also eager to see an update voor the 405V!

thydungeonman, 2023-11-10:
I'd like to see this as well actually.

SuperBadger77, 2023-11-10:
I'd love to see an official release too.

TheGammaSqueeze, 2023-11-22:
Sadly the real life job keeps me very occupied at the moment. 
But the new release will be out in the next week or so.

sammcj, 2023-12-19:
🤞 its out soon ready for Christmas gifting!

rogiervanroon, 2023-12-19:
> 🤞 its out soon ready for Christmas gifting!

You sir, are awesome!

Supermario1967, 2023-12-25:
Hi, I just received my Anbernic RG405V and want to use it with GammaOS from scratch . Is there a date when we can expect the ETA?

Veddy74, 2024-01-03:
I flashed my 505 and it has been working great so on Christmas Eve I flashed my 405V, it worked fine until today when I just get the "G" logo turning at 90 degree angles over and over.  Do I need to "unbrick"?

closed 2024-02-23 by TheGammaSqueeze