TheGammaSqueeze/GammaOS #93: this not is a issue How to Lock again the Bootloader

DarkN-83, 2023-11-20:
first of all i want say the GammaOS project is nice, only what i was not like and i wait for the solution is the screen orientation.

The think is i was back to stock and all the time appears ur bootloader is Unlocked and is little annoying see that every time, and i was start to try things, after looking many things on google, no one talk about that. well after some study i was arrive to the solution... 
how to re-lock the bootloader in the adb console in Windows 11.
– Enter the command: adb reboot bootloader
– Press “Enter” on your keyboard
– Enter the command: fastboot flashing lock
– Press “Enter” on your keyboard
after that reboot the device with the fastboot reboot fastboot and when the fastbootd appears u can reboot easily and check the message and the bootloader is locked again... 
i dont know if works after flash the GammaOS but it works fine when u back from GammaOS to stock firmware and do this steps

TheGammaSqueeze, 2023-11-22:
This is covered in the readme document already :) 
https://github.com/TheGammaSqueeze/GammaOS/blob/main/README.md?plain=1#L190

Thanks for sharing.

closed 2023-11-22 by TheGammaSqueeze

DarkN-83, 2023-12-06:
well i tried the lock after flash the GammaOs and its worked, u can lock the fastboot after flash the system and no have the advice annoying the fastboot is unlocked