TheGammaSqueeze/GammaOS #94: Saturn games fail to load from Daijisho frontend

discus7l, 2023-11-21:
Using both the free and paid version of Yaba Sanshiro 2, the games fail to load from Daijisho. Let me know if any details are required.

veb-xenon, 2023-11-21:
Same here. It just freezes...

dfreivald, 2023-11-22:
It's known bug with Yaba Sanshiro. The YS dev has not updated the emulator for scoped storage, so later versions of android will not work with Daijisho.

Good news is Gamma has a workaround he wants to roll into the next update.

discus7l, 2023-11-22:
Thanks for the heads up! Is it being discussed on a discord channel or something? I'd like to follow the discussion if possible.

TheGammaSqueeze, 2023-11-22:
Yes this is fixed in the next GammaOS release. You can follow any discussions on the Retro Handhelds discord.

closed 2024-02-23 by TheGammaSqueeze