TheGammaSqueeze/GammaOS #105: Psp Games not working

bmakhale, 2023-12-13:
I have 250 PSP games on my 505 and they will not work using multiple emulators. My Saturn games will not either, but I saw that the next update will resolve that.

Jonasbeavis, 2023-12-16:
Are you trying to use CHD files? If so , only PPSSPP dev builds run those , and by experience doesn't run well. Use CSO or ISO only with PS store PPSSPP emulator.

JamesRHarris, 2023-12-16:
On my 505 I had to change the render from vulkan to gl to get PSP working. I noticed that I needed to manually save the config in retroarch to get it to stick.

SqueakyRafiki, 2024-01-02:
Where did you see that the next update will resolve Saturn issues?
 Both the RetroArch cores and the Yaba Sanshiro 2 app is having trouble running Grandia.

closed 2024-02-23 by TheGammaSqueeze