TheGammaSqueeze/GammaOS #106: GammaOS key mapper

ed-fruty, 2023-12-14:
Is the keymapper included in the GammaOS build?

I want to install it, but it's important for my use case to have a key mapper. On the stock OS, it's quite good because it's integrated into the OS without any additional actions or applications.

Is there a video available where I can see what the key mapper in GammaOS looks like ?

david-sigaty, 2024-01-19:
There is not a keymapper built into GammaOS. I've seen some people suggest alternatives like the Mantis app, I have not experimented with that though. Maybe try out a few app alternatives in the stock os to see if you can find one that works for you?

closed 2024-02-23 by TheGammaSqueeze

gbraad, 2024-06-17:
Mantis does not for the Rockchip-based release as drawing over applications is disallowed. not sure if this is the same for this Unisoc based release.